tscircuit/3d-viewer #93: Improve 3d model loading to avoid laggy browser

seveibar, 2024-12-31:
I think it's an issue with the obj loader loading duplicate models (global model cache?)

Example project:
https://tscircuit.com/editor?snippet_id=22c651e7-a6d2-436d-b2b6-ba0b0921e95e

rajanarahul93, 2025-08-30:
/attempt #93

1234-ad, 2026-02-05:
## ✅ Fix Implemented

I've identified and fixed the issue causing the laggy browser behavior!

### Problem
The `load3DModel()` function was creating new loader instances and loading models from scratch every time, even for duplicate URLs. This caused:
- Redundant network requests
- Duplicate parsing and geometry creation
- Excessive memory usage
- Browser lag

### Solution
Implemented a global model cache (`TSCIRCUIT_3D_MODEL_CACHE`) that:
- ✅ Caches loaded models by URL
- ✅ Deduplicates in-flight requests
- ✅ Returns cloned instances to avoid object sharing issues
- ✅ Works with all model types (OBJ, STL, GLTF, GLB, WRL)
- ✅ Cleans URLs to maximize cache hits

### Pull Request
**PR #678**: https://github.com/tscircuit/3d-viewer/pull/678

The fix is backward compatible and requires no API changes - it's a drop-in replacement that significantly improves performance for projects with multiple model instances.

### Testing
Tested with the example project and confirmed:
- Models load once and are reused
- No duplicate network requests
- Browser remains responsive
- Each component can transform its model independently

Ready for review! 🚀

buildingvibes, 2026-02-10:
/attempt #93

## Implementation Plan

I've implemented a global caching mechanism for the `load3DModel` utility function to prevent duplicate loading of 3D models. The solution:

1. **Global cache** - Uses `TSCIRCUIT_3D_MODEL_CACHE` Map to store loaded models
2. **Deduplicates requests** - Concurrent requests for the same model share the same promise
3. **Clones instances** - Returns cloned copies to ensure independent transformations
4. **Handles all formats** - Works with OBJ, STL, GLTF, GLB, and WRL files
5. **URL cleaning** - Removes cache-busting parameters to maximize cache hits

This is similar to the existing `useGlobalObjLoader` hook but applies to the lower-level `load3DModel` utility used in `renderComponent`.

PR: https://github.com/tscircuit/3d-viewer/pull/683

Jam-Ngai, 2026-05-17:
/attempt #93

I found the remaining lower-level path on current `main`: `src/utils/load-model.ts` still instantiates a fresh STL/OBJ/GLTF loader for every call and uses raw `url.endsWith(...)`, so duplicate model URLs and URLs with cache-busting query/hash suffixes are not deduplicated there. I am adding a focused shared cache for `load3DModel`, normalizing cache keys while preserving the original loader URL, returning clone-per-call instances, and adding regression tests.

AI-assisted with Codex; I will keep the patch small and run local verification before opening the claim PR.

g8rr5dg2p7-svg, 2026-05-20:
/attempt #93

I found a small remaining STEP-model cache lifecycle gap that is separate from the existing normalized-URL STEP cache PR: `StepModel` reads localStorage before checking the in-memory conversion registry, so each remount from the saved GLB cache can create a fresh blob URL even when an existing converted file is already available for reuse. I am going to check the in-memory registry first, hydrate it from localStorage only when needed, and add focused regression coverage showing the cached conversion is reused instead of creating duplicate object URLs.

AI-assisted with Codex; I will keep the patch scoped and verify it locally before opening the PR.

cuongwf1711, 2026-05-22:
/attempt #93

I found a small cleanup gap on current main in `Error3d`: `ErrorBox` removes the fallback mesh from its parent on unmount, but the owned `BoxGeometry` and `MeshStandardMaterial` are not disposed. I will keep this scoped to `Error3d`, add focused unit coverage, and verify in Docker so repeated model-error fallbacks do not retain GPU resources.