tscircuit/3d-viewer #94: Hovering over tooltips causes major slowdown on 3d viewer, probably due to "overrendering"

seveibar, 2025-01-01:
https://github.com/user-attachments/assets/9591cc3d-f58d-419b-ac98-9b7912f0e12b

/bounty $5

DrSensor, 2025-01-02:
I suspect the slowdown is not only caused by the tooltip but also the highlighting too.

After quick look, it change the mesh color directly rather than using postprocessing effect.

seveibar, 2025-01-02:
oooh good find @DrSensor !!!!

CC @Abse2001

gavin-openops, 2026-03-01:
/attempt #94

Plan:
1. Reproduce hover slowdown while moving across dense models.
2. Reduce over-rendering by throttling hover position updates in HoverContext (same hovered object).
3. Keep immediate hover/unhover transitions intact.
4. Run format/build/node-bundle checks and open PR with benchmark notes.

closed 2026-03-05 by Abse2001