tscircuit/3d-viewer #115: <Pcb3D /> component

DrSensor, 2025-01-15:
## why?
`<CadViewer />` is good, but you can't swap the renderer, lightning, and so on.
We need to extract the 3d component.
```jsx
<MyCanvas>
  <CustomCamera />
  <WorldEnvironment />
  <Pcb3D>
    …
  </Pcb3D>
</MyCanvas>
```

> [!IMPORTANT]
> required so there is no duplicate dependencies when https://github.com/tscircuit/circuit-to-svg/pull/166 merged
> > I'm using 👇 in that PR
> > ```
> > bun add github:tscircuit/3d-viewer
> > ```
> > so I can use the typescript file directly.
> >
> > [trivia]
> > Q: why not using `bun patch` command?
> > A: patching the package doesn't work because the `./src/**.ts` is missing, it only have `./dist/*.js`

seveibar, 2025-01-15:
Yes that's a good idea, we could have CadViewer be a sort of pre-packaged version with Camera etc. and create a Pcb3D component?

DrSensor, 2025-01-15:
exactly

DrSensor, 2025-01-15:
@ShiboSoftwareDev or @seveibar, do you still remember why you need this?
https://github.com/tscircuit/3d-viewer/blob/51848a0a6679464f9102e8e28625c3740163a377/src/hooks/use-stls-from-geom.ts#L5-L14
https://github.com/tscircuit/3d-viewer/blob/51848a0a6679464f9102e8e28625c3740163a377/src/hooks/use-stls-from-geom.ts#L35-L37

I'm thinking to refactor that so it will be sync instead of async.
see: https://github.com/tscircuit/circuit-to-svg/pull/166/files#diff-0a0a0fcd88aabe21132caeaf2bbf993e67309ddc6514a6d0e5194d0b857a9988

seveibar, 2025-01-15:
oh interesting, my guess is we were using stls for some models, but I can't find any snippets where we use stls currently.

It would appear as `model_stl_url` in `cad_component`

This is an example `cad_component` with an obj URL (this is more common for jlcpcb)

```json
{
  "type": "cad_component",
  "cad_component_id": "cad_component_0",
  "position": {
    "x": -5.1514348342607263e-14,
    "y": 0,
    "z": 0.7
  },
  "rotation": {
    "x": 0,
    "y": 0,
    "z": 0
  },
  "pcb_component_id": "pcb_component_0",
  "source_component_id": "source_component_0",
  "model_obj_url": "https://modelcdn.tscircuit.com/easyeda_models/download?uuid=07c2e528ec9a4008b33211831b7000e1&pn=C7203003&cachebust_origin=https%3A%2F%2Ftscircuit.com"
}
```

seveibar, 2025-01-15:
Oh actually I'm remembering something... I think I had some crazy hack with `jscad-fiber` objects where I converted them to STL and back- sorry my memory isn't super great on the specifics of this code base atm haha but i think you haven't broken anything regarding jscad-fiber if this story still works and puts a random cube on a pcb 😁 https://tscircuit-3d-viewer.vercel.app/?path=/story/cadcomponentwithjscadplan--default